djphazer/O_C-Phazerville #119: Quatermain clock divider

olegchomp, 2025-01-14:
In Quatermain app clock can be divide up to 16, but can it go further (24, 32, 40, 48)? For ex. with Pam workout clock sync is x24 and it will be great to have ability to use same clock in app (especially in small cases).

olegchomp, 2025-01-14:
Also any other app / applet doesn't work with 24 ppqn clock

djphazer, 2025-01-14:
Looks like Quantermain's clock divider parameter has no good reason to be limited to /16... I'll gladly increase the max to maybe /64 (or even /128). I'll take a look at the other stock apps while I'm at it.

Within Hemisphere, you can use the `sync=` parameter in [Clock Setup](https://firmware.phazerville.com/Clock-Setup) to specify the external PPQN, which does go up to 24.

FWIW, you can also set a different PPQN on Pam's for incoming clock if you need to.

olegchomp, 2025-01-14:
Thank you! 

As for hemisphere sync, I've tried to send x24 from Pams and set sync and others parameters, but nothing works. Only way as mention above is to use divider app.

djphazer, 2025-01-14:
For Hemisphere, `sync=` and the internal multipliers/dividers only take effect when the Clock is running. Did you try starting the Clock? Once armed, it will auto-start and auto-stop with TR1 pulses.

olegchomp, 2025-01-24:
> For Hemisphere, `sync=` and the internal multipliers/dividers only take effect when the Clock is running. Did you try starting the Clock? Once armed, it will auto-start and auto-stop with TR1 pulses.

Thank you!

closed 2025-01-24 by olegchomp