djphazer/O_C-Phazerville #121: O_C not working after upgrading to 1.8.3 in DIY 1U O_C (Teensy 3.1/3.2)

s0rtega, 2025-01-20:
Hello @djphazer, 

First of all thank you for developing and maintaining Phazerville, awesome work here. I am having some issues upgrading to 1.8.3 and I wanted to share it here in case you have some ideas or can provide some advice on how to identify where is the issue. 

- **Issue:** after upgrading to 1.8.3, O_C doesn´t boot up but Teensy works normally. It seems that the LCD screen is not working. Rollback to 1.8.2 (any version) fix the issue. 

- **Hardware:** DIY 1U O_C Version with Tensy 3.1/3.2. I believe it´s based in MXMXMX with variable CV (hint: polymorphic CV), but I can't confirm which version is exactly. For more details see the attached picture. 

- **Steps to reproduce & other information:** Simply compile any 1.8.3 build using GitHub options or locally building it will crash the O_C. Tried multiple versions (custom, prod, vor, etc.) with same result. Always working in 1.8.2 without issues. I have compiled one custom 1.8.3 version with the` -DPRINT_DEBUG` flag (Thanks to https://github.com/djphazer/O_C-Phazerville/issues/55) and the O_C seems to boot normally but the screen is off:

```
* O&C BOOTING...
* v1.8.3
Cal.Storage: PAGESIZE=128, PAGES=1, LENGTH=128
Calibration data loaded...
CV scale not set, using default
Set screensaver timeout to 25000
Configuring encoders: normal (0)
* CORE ISR @60us
* UI ISR @1000us
Set screensaver timeout to 25000
```
 
I saw the support for SPI added in the new release (https://github.com/djphazer/O_C-Phazerville/issues/91) so I am wondering if any of the changes could have caused this issue but I prefer to double confirm it with you before going down the wrong way. 

Happy to provide more information / tests if required.

[image]
[image]

djphazer, 2025-01-20:
My apologies - your hardware is probably not handling the display driver changes very well, from commit 893662f0f06f0126fe25627f70a992ae29462c8e

I probably need to revert the sub-page transfer stuff to make it stable again on T3.2.
A potential workaround might be to slow down the screen refresh rate, by increasing the value of REDRAW_TIMEOUT_MS here - https://github.com/djphazer/O_C-Phazerville/blob/phazerville/software/src/OC_config.h#L22
(try something like 10ms)

Also, to clarify, your unit does *NOT* have VOR... the word "polymorphic" is just a fancy way of saying it's a multifunction module ;)

HeWhoWantsJeansAgain, 2025-01-20:
I think I just ran into this flashing the latest version on my o_C builds by CalSynth running T4.0. After flashing with 1.8.3 the display would not come on. I just downgraded to 1.8.2 and they work fine.

s0rtega, 2025-01-21:
Hello there and thanks for the quick feedback. 

>[...] A potential workaround might be to slow down the screen refresh rate, by increasing the value of REDRAW_TIMEOUT_MS here - https://github.com/djphazer/O_C-Phazerville/blob/phazerville/software/src/OC_config.h#L22 (try something like 10ms) [...]
> 

Tried a bunch of values (from 5...20) but the display is still not working. To confirm, I used the [Phazerville-Screen-Capture](https://github.com/PaulStoffregen/Phazerville-Screen-Capture) app. and I can see the screen as expected, so it must be the feature / commit that you pointed out. 

> Also, to clarify, your unit does _NOT_ have VOR... the word "polymorphic" is just a fancy way of saying it's a multifunction module ;)

Ahhh, my bad. Got caught in the _fancy_ language 🗡 .

djphazer, 2025-01-22:
As I'm deciding what to revert, I wonder if just this line would make a difference? "charge pump setting - enable"
https://github.com/djphazer/O_C-Phazerville/blob/phazerville/software/src/src/drivers/SH1106_128x64_driver.cpp#L74

 Which actual display are you all using?

s0rtega, 2025-01-23:
Hello,

> As I'm deciding what to revert, I wonder if just this line would make a difference? "charge pump setting - enable" https://github.com/djphazer/O_C-Phazerville/blob/phazerville/software/src/src/drivers/SH1106_128x64_driver.cpp#L74
> 
> Which actual display are you all using?

"charge pump setting - enable" did the trick. I will test it later tomorrow to see if there are any problem using it normally but I was able to boot up, change apps., move around. It seems totally normal now. 

Here is the display that it is using:

[image]

> I pushed a potential fix, rolling back the display driver changes - 

Also working! Thanks for taking the time looking into this.

djphazer, 2025-01-23:
Awesome, looks like the "charge pump" setting is necessary after all! No clue why I decided to comment it out... probably a "works for me" scenario. That's a simple fix tho, thanks for your help!

djphazer, 2025-02-14:
I pushed a fix here: https://github.com/djphazer/O_C-Phazerville/commit/6da67d2e9d63b0ecc0168b70926ed6e5b2ec6594

...but leaving this open until the next official release build that includes it.

closed 2025-03-06 by djphazer