djphazer/O_C-Phazerville #128: 90 degrees display rotation

twangmodular, 2025-02-11:
Hi,
I have a custom build with 1.3 128x64 OLED which is slim enough to fit 1U. But there is a problem with screen rotation. Can someone push me to the right direction of how to fix this?

[image]

[image]

djphazer, 2025-02-12:
I suggest trying this branch for simplified driver code: https://github.com/djphazer/O_C-Phazerville/commits/revert-large-OLED/

But change line 74 here: https://github.com/djphazer/O_C-Phazerville/blob/revert-large-OLED/software/src/src/drivers/SH1106_128x64_driver.cpp#L74
to `0x020, 0x001,` for vert addressing

twangmodular, 2025-02-12:
> I suggest trying this branch for simplified driver code: https://github.com/djphazer/O_C-Phazerville/commits/revert-large-OLED/
>
> But change line 74 here: [...]

Thanks for reply, Nicholas.
I spent all the day trying to compile the firmware with Arduino IDE. My attempts for today were ended with this error:

**In file included from C:\arduino\O_C-Phazerville-phazerville\software\src\hemisphere_config.h:44,
                 from C:\arduino\O_C-Phazerville-phazerville\software\src\APP_HEMISPHERE.h:44,
                 from C:\arduino\O_C-Phazerville-phazerville\software\src\OC_apps.cpp:46:
C:\arduino\O_C-Phazerville-phazerville\software\src\applets\EbbAndLfo.h:2:10: fatal error: util/util_phase_extractor.h: No such file or directory
    2 | #include "util/util_phase_extractor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: util/util_phase_extractor.h: No such file or directory**

The file util_phase_extractor.h is definitely on its place. I've tried to remove *44#include "applets/EbbAndLfo.h"* in hemisphere_cinfig and other lines from ones mentioned in error message and after that even compiled the .hex file, but with that firmware Teensy has the led indicator off and nothing happens.

I can upload the test sketches. I used all preferences as discribed in instructions. 
I also tried the option with Visual Studio, but I got even more errors while compiling. 

Maybe you can figure what I'm doing wrong. 
Thank you.

djphazer, 2025-02-12:
Apologies, I haven't been testing the builds in Arduino IDE ... that include probably needs to be a relative path: "../util/util_phase_extractor.h"

djphazer, 2025-02-13:
I guess it's a bit more complicated than just the addressing mode... if you're determined, keep reading into the display driver and framebuffer code, and compare with the working u8g2 implementation.