djphazer/O_C-Phazerville #129: Warming up teensy and LM1117

TorkhovRV, 2025-02-23:
After upgrading to version 1.8.3, the LM1117 became very hot, and the teensy warms up a bit. 
I tried different teensy because I have this problem with all of them. Teensy heats up without even being inserted into the module. 
I think it may be related to the frequency of 600 MHz? 
Sorry, I'm a newbie and couldn't build hex file in the platformio.ide and arduino.ide (but blink doesn't heat up teensy)

djphazer, 2025-02-23:
Are you saying this wasn't the case with v1.8.2? Are you using Teensy 4.0 in 8hp hardware?

TorkhovRV, 2025-02-23:
I can't say which version I had before (I don't know), but I think I would have noticed the heating, because I felt the heat from the module panel in the rack. 
Of course, I thought that the lm1117 had failed. But it coincided with the update. I'll try replacing lm1117 to be sure, but why is there heating from usb
Yes teensy 4.0 8hp.
Now I tried to install version 1.8.1, teensy is warm, lm1117 is hot but not very much, I have nothing to measure the temperature with. Hmmm

TorkhovRV, 2025-02-24:
Now I also discovered that when the internal clock is launched, the algorithms do not start (only from the external clock), this is for all versions 1.8. for versions 1.7 everything is fine. 
There are also some fluctuations in the algorithms where inputs are required

djphazer, 2025-02-24:
Did you configure the internal clock multipliers? At some point, I made them all default to OFF after someone complained...

TorkhovRV, 2025-02-24:
> Did you configure the internal clock multipliers? At some point, I made them all default to OFF after someone complained...

I forgot about it :(

djphazer, 2025-02-24:
Back on-topic, I'm not a hardware guy, but...

If you're comparing T4.0 with T3.2, perhaps the T4 is drawing more power? You might be right to suspect the clock speed... slower speeds would probably draw less power. If you have USB connected, have you cut the VUSB power trace for safety? 

Seems like a similar report concluded with faulty components or bad solder joints, but let me try to find that source...

If you can definitively say that things are running hotter in 1.8.3 and not in earlier versions, it could be the display driver change using _slightly_ more CPU... but I doubt that would be significant on Teensy 4.0.

TorkhovRV, 2025-02-24:
I don't have T3.2, but a friend will send it for compare.
I didn't cut the wiring, but I didn't turn on USB and rack power at the same time.
Yes, the difference is noticeable to the touch.
I think I'll try to replace LM1117 with a step-down DC-DC converter (they heat up less) and add a radiator to the teensy.
I'll let you know later

TorkhovRV, 2025-03-05:
[image]
https://github.com/user-attachments/assets/56a5b24c-80a4-4c40-858f-a5f446c93d90

Problem solved.
I replaced LM1117 with a dc-dc converter, now this unit does not heat up. Teensy still heats up but within normal limits, I glued the radiator with heat-conducting glue.
But I still think that the problem is hidden in the teensy frequency work.

closed 2025-03-05 by TorkhovRV

TorkhovRV, 2025-03-11:
I measured the temperatures, and it looks like heating is a normal condition:
T3.2 - 35°C
T4.0 with radiator 40°C
T4.0 without radiator 50°C
Room temperature 28°C