djphazer/O_C-Phazerville #130: TB-3PO transpose input unexpected behaviour.

SungazerExtreme, 2025-02-26:
Not sure if this is a bug or user error but I have a sequence going on the TB-3PO with modulation into the transpose input (the unexpected behaviour occurs no matter what type of modulation). It appears that the sequence keeps getting transposed infinitely until it's only playing one upper note. In my specific patch I have the sequence plugged into the Morphagene's Organise input and it only plays the last slice when the Organise attenuator is at 9pm, below 9pm and it plays back other notes. I can hear the sequence being transposed higher and higher until it's only one note.

Thanks!

djphazer, 2025-02-26:
Strange... have you inadvertently routed the output back to the input?

SungazerExtreme, 2025-02-26:
Not sure if you mean physically patched or internally routed somehow, but it's not physically patched into itself. Not sure about internal routing though or if that's possible?

closed 2025-02-26 by SungazerExtreme

djphazer, 2025-02-26:
Did you figure it out? I suspected internal feedback routing in the Input Mapping...

SungazerExtreme, 2025-02-26:
Yeah I feed stupid now I figured it haha. I had the output of the sequence going into the Make Noise Multimod and back into the transposition input duh. But because the connection was one step removed and because the sequence itself was being modulated by the Multimod before being sent back to transpose, it wasn't happening in an obvious and regular way and it just didn't click.