djphazer/O_C-Phazerville #132: Parameters still accessible while Scope is fullscreen

xmacex, 2025-03-05:
Hello. If I put the _Scope_ application in fullscreen mode, I can still scroll through the various parameters which the apps have with the cursor, and their location is highlighted although value is hidden.

[image]

I would expect these to be hidden and even disabled against messing with them with showing an UI, while the _Scope_ is in fullscreen mode. This in dev/1.9 branch.

It seems that the latest release 1.8.4 works fine.

djphazer, 2025-03-05:
I was aware of this... thanks for the reminder!

xmacex, 2025-03-05:
I was confused that 1.8.4 doesn't have it and works super great. I was under the impression that whatever is in dev/1.9 comes from the main branch, but I must have overlooked the history.

---

Anyway, what a superb firmware... I mean o_C with this firmware – and maybe a Maths – is all one really needs to have a completely satisfying Eurorack for a lifetime.

djphazer, 2025-03-06:
`dev/1.9` is like our staging area for new features... `phazerville` is my "main" branch, considered stable and well-tested for backward compatibility. I didn't cherry-pick the full-screen config UI stuff back into 1.8.4, partially because of this unresolved weirdness.

I'm glad you think so highly of the whole thing! This has been my way of excluding a laptop from my live performance rig... a microcontroller module in a small Eurorack system is well-suited for the duties normally handled by a computer, without worries about latency.

closed 2025-04-01 by djphazer