djphazer/O_C-Phazerville #133: Feature request: Quantermain, restricting output pitch to octave range

Loupi, 2025-03-07:
Hello,

I'm using Quantermain to quantize voltages coming from Joranaloge Orbit3, to produce a bass and melody. However, I cannot find a way to restrict the output pitch to a specific octave range without using an attenuator on the input cv.

I think that the introduction of two new settings, minOct and maxOct, would make this possible.

For example with C3 on first octave, 

minOct: 0
maxOct: 1
C2 would output C3, D2: D3, E2: E3...
C3 would output C3, D3: D3, E3: E3...
C4 would output C4, D4: D4, E4: E4...
C5 would output C4, D5: D4, E5: E4...

minOct: -1
maxOct: 0
C2 would output C2, D2: D2, E2: E2...
C3 would output C3, D3: D3, E3: E3...
C4 would output C3, D4: D3, E4: E3...
C5 would output C3, D5: D3, E5: E3...

This transformation would be applied pre oct and transpose settings.

What do you think about it?

djphazer, 2025-03-07:
I'm not likely to make modifications like this to the stock apps like Quantermain... but I might accept a PR if you or someone else implements it!

I recently added a "note wrap range" parameter to the **Squanch** applet that behaves like you described, wrapping around at a certain range of scale degrees. Combined with Transpose controls, perhaps this could satisfy your use case?

I might later absorb this behavior into the global quantizer engines used by other applets, such as **DualQuant**.

Loupi, 2025-03-07:
Interesting, I'm going to have a look at Squanch.
I also understand your position on modifying stock apps. 

I might give it a shot with a fork/PR, I think this wrapping could be achieved by modifying braids_quantizer Process function. Still the main challenge will be to not break dependant apps.

Thank you!

closed 2025-03-07 by Loupi