djphazer/O_C-Phazerville #136: Piqued adding extra triggers in euclydian mode v1.8.4

cymaticsresearcher, 2025-03-19:
I am sending Piqued a clock from Pams. With "Eucl Length" on 8 and fill on 1 I should only be hearing one trigger every 8 steps but I am hearing 2. When observing the dot that indicates step position it appears one step off from actual fired trigger. 

I re-installed original firmware and this behavior is not present in the Piqued app. 

I then installed v1.8.3 to check if it was happening in that build and replicated the same issue.

djphazer, 2025-03-22:
Thank you for reporting this! The euclidean functions were revised quite a while back, and somehow Piqued was overlooked... there's an off-by-one error with Length.

I'll have a fix ready soon.

djphazer, 2025-03-29:
This is fixed here: https://github.com/djphazer/O_C-Phazerville/commit/7d9bff25978d0a0a969068ca1caaebf8fc96ab4d

Available for Custom Builds, and will be in the next release!

closed 2025-04-01 by djphazer