djphazer/O_C-Phazerville #145: TB-3PO reset/transposition

nellyhsch, 2025-04-10:
Resetting the seed: In earlier versions, I believe prior to 1.8x, when the seed was locked and one of the digits manually changed, a reset pulse updated the seed. Currently, it updates only when the density is changed (at least that's what I found). For me, an update on the reset pulse would make more sense. 

Transposition: Currently, transposing via cv is applied in semitones, previously (also before 1.8x I think), it was done in scale (pre- instead of post-quantizing). Both are valid of course, I would prefer the in-scale transposition. 

(I'm running version 1.9 on a module with a teensy 3.2) 

More importantly though, many thanks for your amazing work!

bennyblancodisco, 2025-04-22:
Please return the In Scale transposition or offer both with a toggle if possible. 

Also, please provide the update to OG Teensy 3.2 hardware crew too.

nellyhsch, 2025-04-28:
Thank you for looking into it - changing the seed immediately would work as well for me as waiting on the reset pulse.
Having the option to change the transposition mode would of course be awesome!

nellyhsch, 2025-07-02:
Great, thank you!