djphazer/O_C-Phazerville #149: Carpeggio with Internal Clock doesn't match external clock behavior

Ericxgao, 2025-05-14:
In the Carpeggio app, if I have internal clock running and assign it as x1 to the applet, it doesn't seem to progress the steps of the applet. However, if I feed in a square wave into trigger 1 it ends up working and cycling through each step of Carpeggio.

Oddly enough, higher clock multiplications do let me step through more of the arpeggio grid, but it will keep resetting to step one early. I have sync off on the internal clock as well.

djphazer, 2025-05-14:
What do you have assigned to the Reset input of each applet?

Ericxgao, 2025-05-14:
> What do you have assigned to the Reset input of each applet?

Ahh I didn't realize there was one. I see that it was on x1 - if I make it some division of the clock it works. Is there a reason why both are set to x1 by default?

djphazer, 2025-05-14:
By default, they should all be set to nothing, `x0` ... unless something strange is going on (which is always a possibility!)

Ericxgao, 2025-05-14:
> By default, they should all be set to nothing, `x0` ... unless something strange is going on (which is always a possibility!)

I see, so it is indeed x0 (clock in and reset are mapped to TR1 and TR2 respectively by default) - however assigning x1 to the applet in the clock setup page will set both reset and clock to x1. Not sure if that's intended. But I also can't think of a more correct default atm, maybe displaying the reset clock divider in the clock setup menu at the same time would be useful.

djphazer, 2025-05-18:
> however assigning x1 to the applet in the clock setup page will set both reset and clock to x1. Not sure if that's intended.
I don't understand / can't reproduce this part. There are two separate multipliers per applet - one for each trigger input. I typically set one to x4 (for Clock) and leave the other as x0 (Reset). Reset happens once automatically whenever you start the Clock.

djphazer, 2025-06-08:
Seems like this was just improper configuration of the internal clock settings. Let me know if there's a real bug I haven't addressed.

closed 2025-06-08 by djphazer