djphazer/O_C-Phazerville #150: BIG OLED 2.4inch does not work with latest firmware.

themangoest, 2025-05-17:
Hello,

Phazerville for T4 and T4.1 does not work for BIG OLED 2.4inch screens.

[image]

I only tested and used these HEX files :

- o_C-phazerville-v1.8.2_T41-f762a08
- o_C-phazerville-v1.8.2_T40-f762a08

I think it doesn't work anymore since 1.8.3 

Can this be checked please?

Thanks,

Gregory

djphazer, 2025-05-17:
My apologies! I reverted the display driver changes for the main releases. You just need to cherry-pick this commit - https://github.com/djphazer/O_C-Phazerville/tree/dev%2Flarge-OLED

I'll rebase it on the latest stuff when I get to my PC later.

djphazer, 2025-05-17:
Rebased and pushed. Latest binaries with large-OLED patch are here: https://github.com/djphazer/O_C-Phazerville/actions/runs/15089176159

The T41 build will use the slower SPI speed for the display if HW_ID voltage is 0.5v or greater, indicating NLM hardware.

djphazer, 2025-05-18:
Just tested it myself... had to push a fix for T40 - https://github.com/djphazer/O_C-Phazerville/actions/runs/15091297186

Sorry for the chaos! XD

themangoest, 2025-05-19:
There seems to be still an issue with the firmware.
Can you share a cooked hex for a t4.1 as a test? Maybe I do something wrong.

[image]

djphazer, 2025-05-19:
Does T40 work? You can also run the T40 firmware on a Teensy 4.1 ...

The T41 build basically expects new 8-channel hardware. It _should_ still run on old hardware, but I haven't been testing it...

biggiesmallsAG, 2025-05-19:
hey @djphazer  - im the guy with the issue Greg posted, just tested T40 and it works (v1.9_T40-f1f1cc2.hex)

djphazer, 2026-09-23:
@themangoest I've rebased the large OLED patches on `main` based on v2.0.1, pushed to the `large-OLED-patch` branch. Download the artifact here: https://github.com/djphazer/O_C-Phazerville/actions/runs/35807412874

That branch only supports T40... if you need it for T32, the same 2 commits can be applied to the latest `T32` branch as well.

As a reminder, you can run the T40 firmware on a Teensy 4.1 for use on O_C hardware, but there is not much to gain from using T4.1 instead of T4.0 on original O_C architecture.