djphazer/O_C-Phazerville #153: WTVCO not showing up in the new 1.9.1

rootapprox, 2025-07-08:
Hello i just upgraded to the new firmware but i dont find the WTVCO applet , i had to revert to 1.9 to find it again , any idea why ? 
Cheers!

djphazer, 2025-07-08:
It's a [secret](https://github.com/djphazer/O_C-Phazerville/commit/8aff67c85776ed2b9a962dd0a8c1c037f219504e) to everybody.... try saying "PEW" three times fast! ⚡

rootapprox, 2025-07-08:
PEWPEWPEW

djphazer, 2025-07-08:
> PEWPEWPEW

lol it's an Easter Egg applet now... what hardware are you trying to run it on?

rootapprox, 2025-07-08:
4.0

rootapprox, 2025-07-08:
@djphazer im sorry for my limited skills regarding this .. i dont know how to do it ..

elkesabe, 2025-08-30:
> et up a

@djphazer Thanks to this answer to rootapprox I manage to dive into Github and fork your project to create my own version of Hemispheres. LOL